OpenIPC/devourer #12: Set _usbTxAggMode to false to solve TX package loss issue

imagemlt, 2025-04-08:
Hello,when I tried to implement TX (https://github.com/imagemlt/DualDevourer) I didn't found that we have already set _usbTxAggMode to True. https://github.com/OpenIPC/devourer/blob/455db503e8ce8bfbd5f5d35ef0f90e7dbb367851/src/HalModule.h#L44 . My implementation is based on PC driver which do not have TX aggregation by default, so this may cause package loss issue. My rtl8812au card is broken these days and I cannot try it these days, I thin you can try turn off this and test if package loss will decrease greatly. Thanks.

imagemlt, 2025-04-08:
I checked https://github.com/svpcom/rtl8812au 
https://github.com/svpcom/rtl8812au/blob/4d45664397fb086f77ecc1604fc605c851d3cb75/hal/rtl8812a/usb/usb_halinit.c#L123
Since only aggregate one package for rtl8812au when turn on this,I think we can just turn it off.

vertexodessa, 2025-04-19:
Thank you @imagemlt , I tested it locally - works just fine

vertexodessa, 2025-04-21:
hi @imagemlt , that's a very interesting idea! 
but if i understand correctly to initialize the card, you first need a kernel driver right?

imagemlt, 2025-04-21:
@vertexodessa https://github.com/imagemlt/wfb-termux-qemu/tree/main/qemu you can see here, in rootfs.img I compiled 8822cu driver for the virtual machine inside qemu. 

[image]

 You can check start.sh and main.cpp for how the qemu is launched and how it works after the qemu virtual machine is killed. Use kernel driver inside qemu to initialize wifi card (configure wifi card to monitor mode), then the qemu is stopped and process outside qemu use libusb_bulk_transfer to read/write data directly from wifi card. With Termux and termux-usb we can run a qemu viritual machine connecting to usb device by libusb without root permission.

vertexodessa, 2025-04-21:
@imagemlt 
yep, I get it
thing is - the main limitation in general Android application is inability to run software as root
that is, unfortunately we cannot install kernel drivers 

I am also implementing a possibility to use different devices in the user space with libusb, but I decided to take a different approach.
I'm compliling the driver as a shared library and then I'm providing functions that are usually provided by the kernel.

this way the application won't require root rights

today I got a very first working prototype for RTL8812EU2, and could receive packets from the drone.

imagemlt, 2025-04-21:
sounds great,looking forward to your work

---- Replied Message ----
| From | Ihor ***@***.***> |
| Date | 04/21/2025 22:08 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [OpenIPC/devourer] Set _usbTxAggMode to false to solve TX package loss issue (Issue #12) |

@imagemlt
yep, I get it
thing is - the main limitation in general Android application is inability to run software as root
that is, unfortunately we cannot install kernel drivers

I am also implementing a possibility to use different devices in the user space with libusb, but I decided to take a different approach.
I'm compliling the driver as a shared library and then I'm providing functions that are usually provided by the kernel.

this way the application won't require root rights

today I got a very first working prototype for RTL8812EU2, and could receive packets from the drone.

—

@imagemlt
yep, I get it
thing is - the main limitation in general Android application is inability to run software as root
that is, unfortunately we cannot install kernel drivers

I am also implementing a possibility to use different devices in the user space with libusb, but I decided to take a different approach.
I'm compliling the driver as a shared library and then I'm providing functions that are usually provided by the kernel.

this way the application won't require root rights

today I got a very first working prototype for RTL8812EU2, and could receive packets from the drone.

closed 2025-05-13 by imagemlt